aicareles/Android-BLE #15: java.lang.NoClassDefFoundError: cn.com.heaton.blelibrary.ble.request.ScanRequest$2

MichaelCyuan, 2018-10-23:
java.lang.NoClassDefFoundError: cn.com.heaton.blelibrary.ble.request.ScanRequest$2
        at cn.com.heaton.blelibrary.ble.request.ScanRequest.<init>(ScanRequest.java:120)
        at java.lang.Class.newInstanceImpl(Native Method)
        at java.lang.Class.newInstance(Class.java:1215)
        at cn.com.heaton.blelibrary.ble.request.Rproxy.init(Rproxy.java:36)
        at cn.com.heaton.blelibrary.ble.proxy.RequestProxy.bindProxy(RequestProxy.java:39)
        at cn.com.heaton.blelibrary.ble.Ble.init(Ble.java:88)
        at cn.com.heaton.blelibrary.ble.Ble.create(Ble.java:98)
        at cn.com.heaton.blelibrary.ble.Ble$Options.create(Ble.java:790)
        at com.gzlzinfo.testble.MainActivity.onCreate(MainActivity.java:38)
        at android.app.Activity.performCreate(Activity.java:5356)
        at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1088)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2338)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2426)
        at android.app.ActivityThread.access$800(ActivityThread.java:158)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1349)
        at android.os.Handler.dispatchMessage(Handler.java:110)
        at android.os.Looper.loop(Looper.java:193)
        at android.app.ActivityThread.main(ActivityThread.java:5390)

ai-curry, 2018-10-23:
好的，非常感谢提出这个问题，这个是低版本（5.0以下）扫描的问题，下个版本更新会解决这个问题

MichaelCyuan, 2018-10-24:
请问我只想通过mac地址连接，但是想用setAutoConnect(true)  setConnectFailedRetryCount(3)该怎么操作，如果断开会自动重连吗

ai-curry, 2018-10-24:
会的，通过mac地址进行连接，库中会新生成一个BleDevice对象（新的对象），setConnectFailedRetryCount(3)这个不需要设置都可以的