aicareles/Android-BLE #30: 怎么设置下，扫描列表只返回支持ble的设备，因为有很多name是空的

shuhao1022, 2019-09-16:
怎么设置下，扫描列表只返回支持ble的设备，因为有很多name是空的

aicareles, 2019-10-09:
拿到结果,过滤下name为空字符串就行了

closed 2019-10-09 by aicareles

shuhao1022, 2019-10-09:
> [Alex-Jerry/Android-BLE] 怎么设置下，扫描列表只返回支持ble的设备，因为有很多name是空的 (#30)Closed #30.
>
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "https://github.com/Alex-Jerry/Android-BLE/issues/30?email_source=notifications\u0026email_token=AMB2BIKGWYY2FYRDN3YRVGLQNVH4DA5CNFSM4IW5IJ4KYY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZUWGYLUNFXW5KTDN5WW2ZLOORPWSZGOUDEYQ7I#event-2697562237",
> "url": "https://github.com/Alex-Jerry/Android-BLE/issues/30?email_source=notifications\u0026email_token=AMB2BIKGWYY2FYRDN3YRVGLQNVH4DA5CNFSM4IW5IJ4KYY3PNVWWK3TUL52HS4DFWZEXG43VMVCXMZLOORHG65DJMZUWGYLUNFXW5KTDN5WW2ZLOORPWSZGOUDEYQ7I#event-2697562237",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]