aicareles/Android-BLE #37: Android 10的连接问题

LdwHly, 2020-01-14:
我的app里，业务是这样的，app启动后，会启动一个定时器，这个定时器会不断的去判断保存在本地的蓝牙设备是否连接上，如果没有连接上，就直接根据保存在本地的mac地址连接（不经过扫描过程），Android 10之前的版本一直都比较顺利，从Android 10出来后，很多用户反馈蓝牙连接不上设备，我这边自己测试Android 10，不断循环一下操作：
连接上了设备之后，重启设备。
多连接十几次后，就连接不上了，然后手机的蓝牙开关可以关闭，但是打不开了，只能重启手机，大神知道这个是什么原因吗？

aicareles, 2020-03-02:
建议更换一台手机测试下,系统蓝牙如果都打不开,很大可能就是系统蓝牙死掉了,这个时候只能重启手机了(并非软件问题)

LdwHly, 2020-03-05:
目前的解决方案是连接前先扫描，扫描到了再去连接就没问题，而之前是不扫描直接根据mac地址去连，多连几次就出现了，pixel 2的手机表现为蓝牙重启不了，而三星手机表现为可以重启，但是还是连接不成功（重启手机就可以连接成功），目前都只是android 10 才有这个表现而已

PainterHu, 2021-09-15:
> 目前的解决方案是连接前先扫描，扫描到了再去连接就没问题，而之前是不扫描直接根据mac地址去连，多连几次就出现了，pixel 2的手机表现为蓝牙重启不了，而三星手机表现为可以重启，但是还是连接不成功（重启手机就可以连接成功），目前都只是android 10 才有这个表现而已

我的Android 10也会出现这个问题，我估计是系统存在的bug，我用的另一个蓝牙框架也是这样，先保存MAC地址在本地，过一个晚上，拿保存在本地的MAC地址重新连接时怎么都连接不上，非得重新去扫描一次，拿到扫描的MAC地址连接才能连上。
请问您找到问题的所在了吗？