aicareles/Android-BLE #43: 关于notify消息接收问题

zhao841359690, 2020-04-22:
通过蓝牙设备发送notify消息，但是手机端接收不到，但是通过nrf测试软件可以接收到消息，消息properties:NOTIFY，是我哪里的配置错了吗？

mBle = Ble.options()//开启配置
                .setLogBleExceptions(true)//设置是否输出打印蓝牙日志（非正式打包请设置为true，以便于调试）
                .setThrowBleException(true)//设置是否抛出蓝牙异常
                .setAutoConnect(false)//设置是否自动连接
                .setFilterScan(true)//设置是否过滤扫描到的设备
                .setConnectFailedRetryCount(3)
                .setConnectTimeout(10 * 1000)//设置连接超时时长（默认10*1000 ms）
                .setScanPeriod(12 * 1000)//设置扫描时长（默认10*1000 ms）
                .setUuidService(UUID.fromString("0000ffe0-0000-1000-8000-00805f9b34fb"))//主服务的uuid
                .setUuidWriteCha(UUID.fromString("0000ffe1-0000-1000-8000-00805f9b34fb"))//可写特征的uuid
                .setUuidReadCha(UUID.fromString("0000ffe2-0000-1000-8000-00805f9b34fb"))//可读特征的uuid
                .setUuidNotify(UUID.fromString("0000ffe2-0000-1000-8000-00805f9b34fb"))
                .create(this);

   mBle.startNotify(device, new BleNotiftCallback<BleDevice>() {}

这是我的代码
UUID是正确的

aicareles, 2020-04-28:
建议你使用connect(device,callbaack)接口进行连接，尽量不要使用address去连接

zhao841359690, 2020-04-29:
而且我二次连接蓝牙的时候不管是读或者写的数据都会执行两次