aicareles/Android-BLE #57: 该库好像没有做心跳检测

chenjiufu95, 2020-08-18:
连接上正常使用，设备断电， 需要等约20s  才会回调 通知APP 断开连接； 需求是  断电后，要很快就知道 该设备已经断开理解？

chenjiufu95, 2020-08-18:
没找到好方法实现 实时断电检测， 用下面代码 自己发送接收 心跳包 判断离线；  对连接上的设备 发送心跳包（每秒），正常使用(连接两台设备)，都能正常接收心跳包，  当测试 将 一台设备断电，  不但断电的设备 不能正常发送心跳包， 没有断电设备也不能发送心跳包， 导致 检测断电（断连接）时， 两个设备都被判定为  断开连接， 没有起到 用心跳包 区分设备是否离线的目的。
 private void SendHeartBeet() {
        for (BleBaseDevice device: mConnectedDeviceMap.values()) {
            boolean ret = readByUuid(device);
            MyLogUtil.e(MyLogUtil.TEST,  (ret ? "True" : "False") + "--" + mConnectedDeviceMap.size() + ", address= " + device.getBleAddress());
        }
    }

 public boolean readByUuid(BleBaseDevice bleDevice) {
        boolean ret = Ble.getInstance().readByUuid(bleDevice, SERVER, CHARACTERISTIC_WRITE, new BleReadCallback<BleDevice>() {
            @Override
            public void onReadSuccess(BleDevice device, BluetoothGattCharacteristic characteristic) {
                super.onReadSuccess(device, characteristic);
                if (mCallBack != null) mCallBack.onReadSuccess(device, characteristic);
                if (mConnectedDeviceMap.containsKey(bleDevice.getBleAddress())) {
                    BleBaseDevice dev = mConnectedDeviceMap.get(bleDevice.getBleAddress());
                    if (dev.getBleAddress().equals(device.getBleAddress())) {
                        dev.setHeartBeatUpdateTime(System.currentTimeMillis());
                        MyLogUtil.e(MyLogUtil.TEST, "=== 心跳更新= " + dev.getHeartBeatStr() + ", address= " + device.getBleAddress());
                    }
                }
            }

            @Override
            public void onReadFailed(BleDevice device, int failedCode) {
                super.onReadFailed(device, failedCode);
                MyLogUtil.e(MyLogUtil.TEST, "=== 读取心跳失败, Code= " + failedCode + ", address= " + device.getBleAddress());
                if (mCallBack != null) mCallBack.onReadFailed(device, failedCode);
            }
        });
        return ret;
    }

chenjiufu95, 2020-08-18:
没找到好方法实现 实时断电检测， 用下面代码 自己发送接收 心跳包 判断离线； 对连接上的设备 发送心跳包（每秒），正常使用(连接两台设备)，都能接收心跳包， 然后将 一台设备断电， 断电的设备 不能正常发送心跳包 正常， 但没有断电设备也不能发送和接收到心跳包， 好像是被阻塞了（读方法的使用不对？），导致检测断电（断连接）时， 两个设备都被判定为 断开连接， 没有起到 用心跳包 区分设备是否离线的目的。
private void SendHeartBeet() {
for (BleBaseDevice device: mConnectedDeviceMap.values()) {
boolean ret = readByUuid(device);
MyLogUtil.e(MyLogUtil.TEST, (ret ? "True" : "False") + "--" + mConnectedDeviceMap.size() + ", address= " + device.getBleAddress());
}
}

public boolean readByUuid(BleBaseDevice bleDevice) {
boolean ret = Ble.getInstance().readByUuid(bleDevice, SERVER, CHARACTERISTIC_WRITE, new BleReadCallback() {
@override
public void onReadSuccess(BleDevice device, BluetoothGattCharacteristic characteristic) {
super.onReadSuccess(device, characteristic);
if (mCallBack != null) mCallBack.onReadSuccess(device, characteristic);
if (mConnectedDeviceMap.containsKey(bleDevice.getBleAddress())) {
BleBaseDevice dev = mConnectedDeviceMap.get(bleDevice.getBleAddress());
if (dev.getBleAddress().equals(device.getBleAddress())) {
dev.setHeartBeatUpdateTime(System.currentTimeMillis());
MyLogUtil.e(MyLogUtil.TEST, "=== 心跳更新= " + dev.getHeartBeatStr() + ", address= " + device.getBleAddress());
}
}
}

        @Override
        public void onReadFailed(BleDevice device, int failedCode) {
            super.onReadFailed(device, failedCode);
            MyLogUtil.e(MyLogUtil.TEST, "=== 读取心跳失败, Code= " + failedCode + ", address= " + device.getBleAddress());
            if (mCallBack != null) mCallBack.onReadFailed(device, failedCode);
        }
    });
    return ret;
}

aicareles, 2020-09-13:
这个如果你们有需要，是要自己去在业务层实现的，并且要硬件设备配合

weiwenhuaming, 2023-12-14:
蓝牙链路层存在心跳 PUD（协议数据单元），在无数据传输时，间隔 60ms 左右发一次空包。
算是一次连接事件，主机-> 从机，从机->主机。可以设置超时时间，如 3 秒，3秒内没有有效空包，就会发起断开。