aicareles/Android-BLE #60: BleNotifyCallback 请问回调为啥不执行？？？？？

lxfwelldone, 2020-09-10:


aicareles, 2020-09-13:
初始化时配置正确的主服务uuid，或者通过enableNotifyByUuid方式打开

lxfwelldone, 2020-09-15:
日志里是提示uuid设置成功的，onChanged，onNotifySuccess， 这两个回调方法都没有执行

`2020-09-15 09:20:59.923 21312-21379/com.example.admin.mybledemo I/AndroidBLE: [3763] BleRequestImpl: service_uuid is set up successfully:226c0000-6476-4566-7562-66734470666d

2020-09-15 09:20:59.924 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] BleRequestImpl: characteristic_uuid: 226c00aa-6476-4566-7562-66734470666d

2020-09-15 09:20:59.925 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] BleRequestImpl: characteristic properties is write

2020-09-15 09:20:59.926 21312-21379/com.example.admin.mybledemo I/AndroidBLE: [3763] BleRequestImpl: write characteristic set up successfully:226c00aa-6476-4566-7562-66734470666d

2020-09-15 09:20:59.927 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] BleRequestImpl: characteristic_uuid: 226c00bb-6476-4566-7562-66734470666d

2020-09-15 09:20:59.928 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] BleRequestImpl: characteristic properties is notify

2020-09-15 09:20:59.930 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] ConnectRequest: onReady>>>> CZNZ+0052FEA6

2020-09-15 09:20:59.931 21312-21312/com.example.admin.mybledemo D/lxf: onReady: 

2020-09-15 09:20:59.932 21312-21312/com.example.admin.mybledemo D/BluetoothGatt: setCharacteristicNotification() - uuid: 226c00bb-6476-4566-7562-66734470666d enable: true

2020-09-15 09:20:59.933 21312-21312/com.example.admin.mybledemo D/BluetoothGatt: writeDescriptor() - uuid: 00002901-0000-1000-8000-00805f9b34fb

2020-09-15 09:20:59.935 21312-21312/com.example.admin.mybledemo D/AndroidBLE: [1] BleRequestImpl: setCharacteristicNotificationInternal is true

2020-09-15 09:20:59.935 21312-21312/com.example.admin.mybledemo D/BluetoothGatt: writeDescriptor() - uuid: 00002902-0000-1000-8000-00805f9b34fb

2020-09-15 09:20:59.936 21312-21312/com.example.admin.mybledemo D/AndroidBLE: [1] BleRequestImpl: setCharacteristicNotificationInternal is true

2020-09-15 09:20:59.936 21312-21312/com.example.admin.mybledemo D/MyBleWrapperCallback: onReady: 

2020-09-15 09:20:59.950 21312-21379/com.example.admin.mybledemo D/BluetoothGatt: onDescriptorWrite() - Device=A4:C1:38:**:**:** handle=57

2020-09-15 09:20:59.951 21312-21379/com.example.admin.mybledemo D/AndroidBLE: [3763] BleRequestImpl: write descriptor uuid:226c00bb-6476-4566-7562-66734470666d

2020-09-15 09:20:59.972 21312-21379/com.example.admin.mybledemo D/BluetoothGatt: onConnectionUpdated() - Device=A4:C1:38:**:**:** interval=9 latency=99 timeout=1125 status=0`

closed 2020-09-17 by lxfwelldone