aicareles/Android-BLE #77: 关于发送大数据量数据分包的问题

phoebewow, 2021-01-19:
/**
     * 写入大数据量的数据（分包）
     * @param device 蓝牙设备对象
     * @param data 写入的总字节数组（如整个文件的字节数组）
     * @param packLength 每包需要发送的长度
     * @param delay 每包之间的时间间隔
     * @param callback 发送结果回调
     * @deprecated Use {@link Ble#writeEntity(EntityData, BleWriteEntityCallback)} instead.
     */
    public void writeEntity(T device, final byte[]data, @IntRange(from = 1,to = 512)int packLength, int delay, BleWriteEntityCallback<T> callback){
        request.writeEntity(device, data, packLength, delay, callback);
    }

    /**
     * 写入大数据量数据，需要延迟(分包)
     * 自动模式下写入大数据量数据，无需延迟，根据系统底层返回结果进行连续写入(分包)
     * @param entityData 数据实体
     * @param callback 写入回调
     */
    public void writeEntity(EntityData entityData, BleWriteEntityCallback<T> callback){
        request.writeEntity(entityData, callback);
    }

在使用上面那种方式的时候可以很快的就发送完成，并且数据都是正确发送的，可是使用下面这种方式的时候每包之间的间隔就非常久，即使在EntityData里填写的delay为1，还是间隔很久，这是啥原因啊？主要是因为上面那种方式在我这里提示是过时的方法了，想找下原因

aicareles, 2021-02-23:
> /**
> * 写入大数据量的数据（分包）
> * @param device 蓝牙设备对象
> * @param data 写入的总字节数组（如整个文件的字节数组）
> * @param packLength 每包需要发送的长度
> * @param delay 每包之间的时间间隔
> * @param callback 发送结果回调
> * @deprecated Use {@link Ble#writeEntity(EntityData, BleWriteEntityCallback)} instead.
> */
> public void writeEntity(T device, final byte[]data, @IntRange(from = 1,to = 512)int packLength, int delay, BleWriteEntityCallback callback){
> request.writeEntity(device, data, packLength, delay, callback);
> }
> 
> ```
> /**
>  * 写入大数据量数据，需要延迟(分包)
>  * 自动模式下写入大数据量数据，无需延迟，根据系统底层返回结果进行连续写入(分包)
>  * @param entityData 数据实体
>  * @param callback 写入回调
>  */
> public void writeEntity(EntityData entityData, BleWriteEntityCallback<T> callback){
>     request.writeEntity(entityData, callback);
> }
> ```
> 
> 在使用上面那种方式的时候可以很快的就发送完成，并且数据都是正确发送的，可是使用下面这种方式的时候每包之间的间隔就非常久，即使在EntityData里填写的delay为1，还是间隔很久，这是啥原因啊？主要是因为上面那种方式在我这里提示是过时的方法了，想找下原因

检查autoWriteMode属性值是否为true，如果为true则delay不会生效

phoebewow, 2021-07-01:
非常感谢，我试下，这个autoWriteMode默认应该是false吧，我看源码用的是java，默认没填值，按理应该就是false。
我测试这个的时候并没有设置过这个值，或许我再试下用EntityData.Builder().setDelay(10L).setAutoWriteMode(false).setPackLength(Constant.MTU_SIZE).setData(data).build()
这个方式来看看能不能按照我所想的发送吧。。

phoebewow, 2021-07-01:
> 检查autoWriteMode属性值是否为true，如果为true则delay不会生效

 非常感谢，我试下，这个autoWriteMode默认应该是false吧，我看源码用的是java，默认没填值，按理应该就是false。
我测试这个的时候并没有设置过这个值，或许我再试下用EntityData.Builder().setDelay(10L).setAutoWriteMode(false).setPackLength(Constant.MTU_SIZE).setData(data).build()
这个方式来看看能不能按照我所想的发送吧。。

wudouxingjun, 2023-07-12:
> 这种方式发送大量数据，如果指定Write Uuid 如何发送呢？直接发送显示发送失败

wudouxingjun, 2023-07-13:
设置autoWriteMode为FALSE，利用多线程一直发送数据，发现数据发送几个就失败了，请问如何处理

wudouxingjun, 2023-07-13:
2023-07-13 15:39:01.340 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.340 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.340 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.37
2023-07-13 15:39:01.341 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.346 32617-32617 VRI[DeviceInfoActivity] com.example.admin.mybledemo          D  onFocusEvent true
2023-07-13 15:39:01.347 32617-32617 VRI[DeviceInfoActivity] com.example.admin.mybledemo          D  send msg MSG_WINDOW_FOCUS_CHANGED with caller android.view.ViewRootImplExtImpl.markAndDumpWindowFocusChangeMsg:956 android.view.ViewRootImpl.windowFocusChanged:9837 android.view.ViewRootImpl$WindowInputEventReceiver.onFocusEvent:9550 android.os.MessageQueue.nativePollOnce:-2 android.os.MessageQueue.next:349 android.os.Looper.loopOnce:186 android.os.Looper.loop:351 android.app.ActivityThread.main:8416 java.lang.reflect.Method.invoke:-2 com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run:584 com.android.internal.os.ZygoteInit.main:1013 <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> <bottom of call stack> 
2023-07-13 15:39:01.347 32617-32617 VRI[DeviceInfoActivity] com.example.admin.mybledemo          D  handleWindowFocusChanged mWindowFocusChanged true mUpcomingWindowFocus true mAdded true
2023-07-13 15:39:01.348 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.348 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.74
2023-07-13 15:39:01.349 32617-4835  AndroidBLE              com.example.admin.mybledemo          D  [20474] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.349 32617-4835  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.357 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.357 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 1.0
2023-07-13 15:39:01.357 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.357 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.362 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] writeEntity: onWriteSuccess
2023-07-13 15:39:01.363 32617-32617 Compatibil...geReporter com.example.admin.mybledemo          D  Compat change id reported: 147798919; UID 10177; state: ENABLED
2023-07-13 15:39:01.367 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.367 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.367 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.368 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.37
2023-07-13 15:39:01.376 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.377 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.377 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.378 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.74
2023-07-13 15:39:01.386 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.386 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 1.0
2023-07-13 15:39:01.389 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.389 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.392 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] writeEntity: onWriteSuccess
2023-07-13 15:39:01.396 32617-5074  AndroidBLE              com.example.admin.mybledemo          D  [20478] BleRequestImpl: 54:3D:21:91:13:BC -- write result:true
2023-07-13 15:39:01.396 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.37
2023-07-13 15:39:01.397 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
[... cut ...]
2023-07-13 15:39:01.500 32617-5074  writeEntity             com.example.admin.mybledemo          D  当前发送进度: 0.37
2023-07-13 15:39:01.500 32617-4766  AndroidBLE              com.example.admin.mybledemo          D  [20462] BleRequestImpl: 54:3D:21:91:13:BC-----write success----- status: 0
2023-07-13 15:39:01.500 32617-4766  MyBleWrapperCallback    com.example.admin.mybledemo          D  onWriteSuccess: 
2023-07-13 15:39:01.505 32617-5074  writeEntity             com.example.admin.mybledemo          D  onWriteCancel: 
以上是用多线程一直发送数据的结果，传输几次数据后自动取消，是因为堆栈的问题吗日志没有看出来