aicareles/Android-BLE #84: 系统可搜索到蓝牙，startScan 一个都搜不到，权限都开了，加长了时间都搜索不到

chenni521, 2021-07-16:
I/chatty: uid=10173(com.swaddling.care) identical 11 lines
2021-07-16 09:48:53.451 20043-20043 I/AndroidBLE: [1] BluetoothScannerImplLol: onScanResult: null
2021-07-16 09:48:53.457 20043-20043 I/AndroidBLE: [1] BluetoothScannerImplLol: onScanResult: null
2021-07-16 09:48:54.085 20043-20043 I/chatty: uid=10173(com.swaddling.care) identical 38 lines
2021-07-16 09:48:54.089 20043-20043 I/AndroidBLE: [1] BluetoothScannerImplLol: onScanResult: null
2021-07-16 09:48:54.112 20043-20043 I/AndroidBLE: [1] BluetoothScannerImplLol: onScanResult: null
2021-07-16 09:48:54.956 20043-20043 I/chatty: uid=10173(com.swaddling.care) identical 47 lines
2021-07-16 09:48:54.970 20043-20043 I/AndroidBLE: [1] BluetoothScannerImplLol: onScanResult: null
2021-07-16 09:48:54.987 20043-20043 D/BluetoothAdapter: isLeEnabled(): ON
2021-07-16 09:48:59.467 20043-20043 E/AndroidBLE: [1] : 系统蓝牙已关闭

zhuxiaoxue, 2021-08-03:
多半是没有开启全局定位开关，可以检查一下

chenni521, 2021-08-03:
你好已经解决，是硬件的问题，------------------&nbsp;原始邮件&nbsp;------------------
***@***.***&gt;
发送时间:&nbsp;2021年8月3日(星期二) 下午5:35
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [aicareles/Android-BLE] 系统可搜索到蓝牙，startScan 一个都搜不到，权限都开了，加长了时间都搜索不到 (#84)

closed 2025-02-18 by aicareles