aicareles/Android-BLE #104: 您好，BleRssiDevice 对象能支持转换为BluetoothDevice吗？

PPLLock, 2022-09-26:
您好，感谢分享这个框架，BleRssiDevice 对象能支持转换为BluetoothDevice吗？我们接入第三方蓝牙SDK，需要传入BluetoothDevice对象类型完成初始化，我看到库中所有的扫描和连接对象都是BleRssDevice

StrikeZXA, 2024-08-31:
我也和你查找类似的方法，但也找不到，基于3.3.1版本直接修改了源码    

下载框架后，在自己工程引入core，然后按下面修改即可

类BleWrapperCallback的onLeScan方法修改成如下
`@Override
public void onLeScan(BluetoothDevice originalDevice, T device, int rssi, byte[] scanRecord) {
}`

类ScanRequest修改成如下
`    @Override
    public void onLeScan(BluetoothDevice device, int rssi, byte[] scanRecord) {
        if (device == null) return;
        String address = device.getAddress();
        T bleDevice = getDevice(address);
        if (bleDevice == null) {
            bleDevice = (T) Ble.options().getFactory().create(address, device.getName());
            bleDevice.setDeviceType(device.getType());
            if (bleScanCallback != null) {
                bleScanCallback.onLeScan(device,bleDevice, rssi, scanRecord);
            }
            if (bleWrapperCallback != null){
                bleWrapperCallback.onLeScan(device,bleDevice, rssi, scanRecord);
            }
            scanDevices.put(device.getAddress(), bleDevice);
        } else {
            if (!Ble.options().isIgnoreRepeat) {//无需过滤
                if (bleScanCallback != null) {
                    bleScanCallback.onLeScan(device,bleDevice, rssi, scanRecord);
                }
                if (bleWrapperCallback != null){
                    bleWrapperCallback.onLeScan(device,bleDevice, rssi, scanRecord);
                }
            }
        }
    }`

类BleScanCallback
` public abstract void onLeScan(BluetoothDevice originalDevice, T device, int rssi, byte[] scanRecord);`

aicareles, 2025-02-23:
> 您好，感谢分享这个框架，BleRssiDevice 对象能支持转换为BluetoothDevice吗？我们接入第三方蓝牙SDK，需要传入BluetoothDevice对象类型完成初始化，我看到库中所有的扫描和连接对象都是BleRssDevice

可以使用onServicesDiscovered(BleDevice device, BluetoothGatt gatt)，通过gatt.getDevice获取

closed 2025-02-23 by aicareles