aicareles/Android-BLE #106: startScan报错：UndeclaredThrowableException

baowenqing, 2022-11-02:
java.lang.RuntimeException: Unable to start activity ComponentInfo{com.rct.ros/com.rct.ros.ui.main.activity.DefaultPrinterActivity}: java.lang.reflect.UndeclaredThrowableException
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2666)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2727)
        at android.app.ActivityThread.-wrap12(ActivityThread.java)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1478)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:154)
        at android.app.ActivityThread.main(ActivityThread.java:6121)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:892)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:782)
     Caused by: java.lang.reflect.UndeclaredThrowableException
        at $Proxy1.startScan(Unknown Source)

f262866551, 2023-08-03:
ble.released();会出这个

Wodner, 2024-04-25:
解决了吗

Jamishon, 2024-08-11:
I met the same problem .

 java.lang.RuntimeException: Unable to start activity ComponentInfo{cn.wywk.yiot/cn.wywk.yiot.device.bluetooth.BleActivity}: java.lang.reflect.UndeclaredThrowableException
                                                                                                    	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3993)
                                                                                                    	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:4173)
                                                                                                    	at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:114)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeNonLifecycleItem(TransactionExecutor.java:231)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:152)
                                                                                                    	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:93)
                                                                                                    	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2595)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:107)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:232)
                                                                                                    	at android.os.Looper.loop(Looper.java:317)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:8592)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:580)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:878)
                                                                                                    Caused by: java.lang.reflect.UndeclaredThrowableException
                                                                                                    	at $Proxy3.startScan(Unknown Source)
                                                                                                    	at cn.com.heaton.blelibrary.ble.Ble.startScan(Ble.java:157)