aicareles/Android-BLE #107: 链接成功后无法设置通知

15239545925, 2022-11-15:
1.设置通知后回调方法不会走，以下代码里面的回调都不会走：
private BleNotifyCallback notifyCallback = new BleNotifyCallback<BleDevice>() {
        @Override
        public void onChanged(BleDevice device, BluetoothGattCharacteristic characteristic) {
            UUID uuid = characteristic.getUuid();
            BleLog.e(TAG, "onChanged==uuid:" + uuid.toString());
            BleLog.e(TAG, "onChanged==data:" + ByteUtils.toHexString(characteristic.getValue()));
            runOnUiThread(new Runnable() {
                @Override
                public void run() {
                    showToast(String.format("收到设备通知数据: %s", ByteUtils.toHexString(characteristic.getValue())));
                }
            });
        }

        @Override
        public void onNotifySuccess(BleDevice device) {
            super.onNotifySuccess(device);
            BleLog.e(TAG, "onNotifySuccess: " + device.getBleName());
        }

        @Override
        public void onNotifyCanceled(BleDevice device) {
            super.onNotifyCanceled(device);
            BleLog.e(TAG, "onNotifyCanceled: 取消通知");
        }

        @Override
        public void onNotifyFailed(BleDevice device, int failedCode) {
            super.onNotifyFailed(device, failedCode);
            BleLog.e(TAG, "onNotifyFailed: " + device.getBleName() + "错误码" + failedCode);
        }
    };
2.但是源码输出的日志：像是打开通知了，但是收不到设备返回的数据
 D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe2-0000-1000-8000-00805f9b34fb enable: tr
 D/AndroidBLE: [2] BleRequestImpl: setCharacteristicNotificationInternal is true
 D/AndroidBLE: [2] BleRequestImpl: setCharacteristicNotificationInternal is true
 D/AndroidBLE: [17556] BleRequestImpl: write descriptor uuid:0000ffe2-0000-1000-8000-00805f9b34fb
 D/AndroidBLE: [17556] BleRequestImpl: 80:A0:36:BF:29:39-----write success----- status: 0
 E/MainActivity: onWriteSuccess -- 发送成功:
 D/AndroidBLE: [2] BleRequestImpl: 80:A0:36:BF:29:39 -- write result:true
3、我用其他的蓝牙调试助手是可以收到的，但是这个框架我无法接受

JsonCooler, 2023-04-12:
库的问题,今天我找了 特么两天问题，我以为我代码问题，我看demo又能打开通知，最后把demo的代码对比了下远程库，发现BleRequestImpl类里面多了一个变量 打开通知的时候对这个变量判断了，要么不能打开通知要么多个特征值的时候只会打开第一个通知

Jasongq, 2023-05-23:
我是用远程仓库依赖的代码，可以开启通知，用本地的demo里的开启订阅通知失败，发现BleRequestImpl类里notifyIndex 变量远程仓库的没注释，可以开启订阅通知，但是本地的代码里注释了，换了一种写法，有些蓝牙就无法开启通知

closed 2025-06-25 by aicareles