aicareles/Android-BLE #112: 下载该库源码自行二次定制使用是否需要您的授权

walkeZ, 2023-04-13:
留意到有版权证书：
/*
 * Copyright (C)  aicareles, Android-BLE Open Source Project
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *      http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

下载该库源码自行二次定制使用是否需要您的授权？如果需要怎么办理？

aicareles, 2023-08-03:
> 留意到有版权证书： /*
>
> * Copyright (C)  aicareles, Android-BLE Open Source Project
> * Licensed under the Apache License, Version 2.0 (the "License");
> * you may not use this file except in compliance with the [...]

如果你自己开发的话不需要我授权，可以免费使用，也可以找我进行二次定制开发。

walkeZ, 2023-08-10:
好的，谢谢，我自己进行2次开发

发自我的iPhone

  

  
 
 
 
 
  http://www.apache.org/licenses/LICENSE-2.0

closed 2024-07-12 by aicareles