aicareles/Android-BLE #120: 使用kotlin编写，报错java.lang.NullPointerException: Parameter specified as non-null is null: 。

hwhjxjs, 2024-09-03:
`Ble.options() //开启配置
            .setLogBleEnable(true) //设置是否输出打印蓝牙日志（非正式打包请设置为true，以便于调试）
            .setThrowBleException(true) //设置是否抛出蓝牙异常 （默认true）
            .setAutoConnect(false) //设置是否自动连接 （默认false）
            .setIgnoreRepeat(false) //设置是否过滤扫描到的设备(已扫描到的不会再次扫描)
            .setConnectTimeout((10 * 1000).toLong()) //设置连接超时时长（默认10*1000 ms）
            .setMaxConnectNum(7) //最大连接数量
            .setScanPeriod((12 * 1000).toLong()) //设置扫描时长（默认10*1000 ms）
            .setUuidService(UUID.fromString(UuidUtils.uuid16To128("fd00"))) //设置主服务的uuid（必填）
            .setUuidWriteCha(UUID.fromString(UuidUtils.uuid16To128("fd01"))) //设置可写特征的uuid （必填,否则写入失败）
            .setUuidReadCha(UUID.fromString(UuidUtils.uuid16To128("fd02"))) //设置可读特征的uuid （选填）
            .setUuidNotifyCha(UUID.fromString(UuidUtils.uuid16To128("fd03"))) //设置可通知特征的uuid （选填，库中默认已匹配可通知特征的uuid）
            .setFactory(object : BleFactory<BleDeviceCustom?>() {
                //实现自定义BleDevice时必须设置
                override fun create(address: String, name: String): BleDeviceCustom {
                    return BleDeviceCustom(address, name) //自定义BleDevice的子类
                }
            })
            .create<BleDeviceCustom>(this, object : InitCallback {
                override fun success() {
                    BleLog.e("Wapplication", "初始化成功")
                }

                override fun failed(failedCode: Int) {
                    BleLog.e("Wapplication", "初始化失败：$failedCode")
                }
            })`

自定义model会报空指针错误，使用BleDevice一切正常，跟踪已经初始化完成 但是scan的时候报空指针

liyunfu, 2025-01-09:
兄弟，你解决了吗

aicareles, 2025-01-23:
> `Ble.options() //开启配置 .setLogBleEnable(true) //设置是否输出打印蓝牙日志（非正式打包请设置为true，以便于调试） .setThrowBleException(true) //设置是否抛出蓝牙异常 （默认true） .setAutoConnect(false) //设置是否自动连接 （默认false） .setIgnoreRepeat(false) //设置是否过滤扫描到的设备(已扫描到的不会再次扫描) .setConnectTimeout((10 * 1000).toLong()) //设置连接超时时长（默认10_1000 ms） .setMaxConnectNum(7) //最大连接数量 .setScanPeriod((12 * 1000).toLong()) //设置扫描时长（默认10_1000 ms） .setUuidService(UUID.fromString(UuidUtils.uuid16To128("fd00"))) //设置主服务的uuid（必填） .setUuidWriteCha(UUID.fromString(UuidUtils.uuid16To128("fd01"))) //设置可写特征的uuid （必填,否则写入失败） .setUuidReadCha(UUID.fromString(UuidUtils.uuid16To128("fd02"))) //设置可读特征的uuid （选填） .setUuidNotifyCha(UUID.fromString(UuidUtils.uuid16To128("fd03"))) //设置可通知特征的uuid （选填，库中默认已匹配可通知特征的uuid） .setFactory(object : BleFactory<BleDeviceCustom?>() { //实现自定义BleDevice时必须设置 override fun create(address: String, name: String): BleDeviceCustom { return BleDeviceCustom(address, name) //自定义BleDevice的子类 } }) .create(this, object : InitCallback { override fun success() { BleLog.e("Wapplication", "初始化成功") }
> 
> ```
>             override fun failed(failedCode: Int) {
>                 BleLog.e("Wapplication", "初始化失败：$failedCode")
>             }
>         })`
> ```
> 
> 自定义model会报空指针错误，使用BleDevice一切正常，跟踪已经初始化完成 但是scan的时候报空指针

这个问题下个版本优化，你先用BleDevice，里面有扩展字段，可以支持你自定义